iRobotEducation/create3_examples #14: FeatureRequest: Coverage Server should log dock and undock actions

slowrunner, 2021-11-02:
Currently the Coverage Server logs "Accepting goal request" when passed a proper explore goal, and continues to log actions while performing the explore goal.

The Coverage Server does not log anything for either the /dock or /undock goals.  It at least should log accepting these goals (WIBNI the Coverage Server logged what goal was accepted), and success or failure.

alsora, 2021-11-02:
The dock and undock actions commanded by the coverage server are implementation details.
I think it's better to just print that a behavior is starting, as you described in https://github.com/iRobotEducation/create3_examples/issues/11

slowrunner, 2021-11-02:
no problem with that - loving anything you all can get to.  I'm just having fun playing with you all's code.

slowrunner, 2021-11-03:
I see my confusion now - the /undock and /dock goals (and wall follow) are not going to the coverage server at all  - they are subscribed to by the bot/sim - so the Coverage server will not display "Accepting goal" for them.   

Sorry.

closed 2021-11-03 by slowrunner