iRobotEducation/create3_examples #24: create3_coverage fails to build

arielc-brillianetor, 2022-06-02:
Hi,
Running the build on the create3_coverage project fails. the output of the error states that the create3_coverage_core.lib file doesn't exist. 
I made changes in the CMakeLists.exe file which solved the problem but I'm not sure that the change was correct.
The original file included:
"add_library(${library_name} **SHARED**
    src/behaviors/dock-behavior.cpp
    src/behaviors/drive-straight-behavior.cpp
    src/behaviors/reflex-behavior.cpp
    src/behaviors/rotate-behavior.cpp
    src/behaviors/spiral-behavior.cpp
    src/behaviors/undock-behavior.cpp
    src/behaviors/utils.cpp
    src/coverage_state_machine.cpp
    src/create3_coverage_node.cpp
)"

I changed the SHARED to STATIC so:
add_library(${library_name} **STATIC**
    src/behaviors/dock-behavior.cpp
    src/behaviors/drive-straight-behavior.cpp
    src/behaviors/reflex-behavior.cpp
    src/behaviors/rotate-behavior.cpp
    src/behaviors/spiral-behavior.cpp
    src/behaviors/undock-behavior.cpp
    src/behaviors/utils.cpp
    src/coverage_state_machine.cpp
    src/create3_coverage_node.cpp
)
Building then was successful and the create3_coverage_core.lib file was created. Then I changed it back to SHARED and built it again and it worked.
Thanks for your help,
Ariel

alsora, 2022-06-02:
Hi Ariel,
this change is not correct.
Whether the library is shared or static does not have any impact on finding it.

If you could post more details about your system, the commands that you are running and the complete output of the build step, we can help you figure out the real problem.

shamlian, 2022-06-02:
@arielc-brillianetor reported they are in ROS 2 on Windows and without rosdep in #25 -- Ariel, can you provide more on what exactly you're running and the full output, please? We'd like to help.