iRobotEducation/create3_examples #25: create3_teleop missing project

arielc-brillianetor, 2022-06-02:
Hi,
The create3_teleop project is missing a reference to the ROS2 teleop_twist_keyboard project.
It would be helpful if a link of the project was added to the Readme file as well as adding it to the installations setup.
Thanks a lot,
Ariel

alsora, 2022-06-02:
Hi,
the `create3_teleop` project is listing this dependency in its package.xml https://github.com/iRobotEducation/create3_examples/blob/galactic/create3_teleop/package.xml#L12-L13

I just tested this on a clean system and this dependency is correctly installed when running `rosdep install --from-path src --ignore-src -yi` as per the build instructions in the README.md

arielc-brillianetor, 2022-06-02:
Hi,
Sorry for not adding, but I'm running ROS2 on windows, and no rosdep. For me the project didn't install.
Thanks

shamlian, 2022-06-02:
Sounds to me like the problem you are having is with rosdep, not with our examples. It seems like this is a [known issue](https://github.com/ros-infrastructure/rosdep/issues/733). I can add a note to the readme that if a user has not or cannot install rosdep, that they should install the depenedencies manually, but that should go without saying -- in general, if the steps in a project's parent README.md are not followed, it's unfair to expect children to function properly out of the box.

closed 2022-06-02 by shamlian