iRobotEducation/create3_examples #31: create3_coverage robot namespace not supported

JonathanNash21, 2022-09-16:
I'm trying to use the create3_coverage command with a robot that I created a unique namespace for (I intend to use multiple robots on the same network in the future). However, I get an output saying that "some subscriptions haven't discovered their publishers yet" and "robot nodes haven't been discovered yet". 

[image]

I suspect this is because of the namespace, and the create3_coverage works on a no-namespace assumption (see below image, the /create3_coverage node is isolated and every other node is tied to /r1/..., where r1 is the namespace of my robot).

[image]

I tried editing the create3_coverage_node.cpp file to add in "/r1/" before the rclcpp::Node creation and all of the m_... actions, params, etc, and then re-building the project, but the changes didn't do anything. Is there something else I should be doing to let create3_coverage attach to my robot's namespace? Thank you.

alsora, 2022-09-18:
Hi @JonathanNash21, there isn't any "no namespace assumption" in the coverage example.

As you pointed out, if your robot is running in a namespace you will also have to either run the coverage example in the same namespace or remap all the publications and subscriptions to the namespaced names.

The quickest solution is definitely to start the coverage node in the same namespace as the robot.

```
ros2 run create3_coverage create3_coverage --ros-args -r __ns:=/r1
```

See for details https://docs.ros.org/en/galactic/How-To-Guides/Node-arguments.html

JonathanNash21, 2022-09-19:
Appreciate the helpful response, thank you!

closed 2022-09-19 by JonathanNash21