iRobotEducation/create3_examples #35: Simpe exploration without Lidar

carlitoselmago, 2022-12-08:
I was wondering if there could be an example of a basic aproach to exploration, something like SLAM but without lidar? just something that works with the default sensors of create3 (bump, ir)

slowrunner, 2024-03-08:
@carlitoselmago DISCLAIMER:  I'm not affiliated with iRobotEducation

My thoughts on this are here  https://github.com/iRobotEducation/create3_docs/discussions/567#discussion-6342564

carlitoselmago, 2024-03-09:
I continued working on this some months ago, and got some interesting results using a basic algorithm of curiosity-driven AI based on a video camera. In this aproach, the sense of "where to go" was ultimately driven by getting bored of what has been seen already. Making my robot trying to escape the enclosed space all the time and building a fascination on doors as "gates to the unknown" which in my case meant going outside on the street.