iRobotEducation/create3_examples #43: iRobot Create3 Dance tutorial not working

bluepra, 2023-02-22:
Setup:
Raspberry Pi 4
Ubuntu 22.04
ROS 2 Humble installed

I have the Pi connected to the Roomba's Create® 3 adapter board via a USB-C to USB-C cable.

I am following this instructions:
https://github.com/iRobotEducation/create3_examples/tree/humble/create3_examples_py

This is the command I am running (according to the README): 
```ros2 run create3_examples_py create3_dance```

The Roomba does not move at all, and the terminal has this message showing up:
```[dance_command_publisher] service not available, waiting again```

What could be causing this issue? Any help is appreciated!

carlsondc-ceva, 2023-02-23:
I'd start with a `ros2 service list` and `ros2 node list` to see if the service in question (probably [this one](https://github.com/iRobotEducation/create3_examples/blob/humble/create3_examples_py/create3_examples_py/dance/dance_choreograph.py#L119) ) is visible. If it's not, then I'd check a couple things:

1. Did you make the changes [described here](https://iroboteducation.github.io/create3_docs/setup/pi4/) to usercfg.txt, cmdline.txt, and network-config to enable ethernet over USB?
2. Is the switch on the create3 set to USB or bluetooth?
3. Are you using the same middleware and ROS_DOMAIN_ID as the create3 (as configured through the web server)?
4. Did you configure the create3 to use a namespace? create3_dance expectes it to be in the root namespace.

carlsondc-ceva, 2023-02-24:
you should be able to see a bunch of other nodes from the create3 platform. I'm still using galactic (firmware G.4.3) so things may differ a bit, but I see the following when I'm not running anything on the pi:

```
/_internal/composite_hazard
/_internal/kinematics_engine
/_internal/mobility
/_internal/stasis
/mobility_monitor
/motion_control
/robot_state
/static_transform
/system_health
/ui_mgr
```

If you connect to the pi in a web browser, is the RMW_IMPLEMENTATION set to `rmw_cyclonedds_cpp`? 

If you don't make the changes to usercfg.txt, cmdline.txt, and network-config then you will only be able to talk to the create3 over wifi (not USB). That should be fine as long as your wifi network allows multicast. I ran into problems at one point in my office because multicast traffic was only visible to devices that were connected to the same access point -- so if my pi and my create 3 happened to connect to two different AP's, they could ping each other but not discover each others' ROS nodes.

bluepra, 2023-02-28:
I am trying to communicate from my Pi to the Roomba via a direct USB C connection. I have configured the USB0 (USB C) port on my Pi to act as an ethernet connection. 
From my pi when I say ```ping 192.168.186.2``` I get a successful connection. This tells me that the Pi can talk to the Roomba (correct?). 
I also can go to this IP 192.168.186.2 from my browser and successfully connect to and restart the Roomba.
Lastly, I am using the rmw_fastrtps_cpp, and my Roomba also has the same Middleware. 

I configured the fastDDS on the Roomba to connect to 192.168.186.2:11811 and copied the same settings into the FastDDS XML configuration file. However, ```ros2 topic list``` still does not provide the full list of topics it should have access to.

shamlian, 2023-02-28:
> I configured the fastDDS on the Roomba to connect to 192.168.186.2:11811 and copied the same settings into the FastDDS XML configuration file. However, `ros2 topic list` still does not provide the full list of topics it should have access to.

Is there a reason you've used the Fast-DDS discovery server? If there are only two nodes here directly connected, I think that will cause problems. If instead you are using the Pi to get the Robot onto the Internet (and to connect to your computer), I think you may have to configure that computer as a superclient? I'm not 100% sure but let me see if I can find an expert.

shamlian, 2023-02-28:
Also -- I'm confused -- you said you're using Fast-DDS but you also said you have `export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp` in your .bashrc. Which is it?

bluepra, 2023-03-03:
So I tried running the dance demo on 2 different Operating Systems and ROS2 versions.

### Ubuntu 22.04 64 bit- Desktop
* I am using ROS2 Humble and I followed these instructions to install https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html which matches https://iroboteducation.github.io/create3_docs/setup/ubuntu2204/ pretty closely.
* I have tried both ```RMW_IMPLEMENTATION=rmw_cyclonedds_cpp``` and ```RMW_IMPLEMENTATION=rmw_fastrtps_cpp``` but neither seem to make the desired result work. 
* Like I mentioned, when I do ```ping 192.168.186.2``` I get a successful connection with the Roomba, so I  USB connection is likely not an issue. 
* ```ros2 topic list``` still does not provide the full list of topics it should have access to.
* I only tried Fast-DDS discovery server to see if it would make a difference.

### Ubuntu 20.04 64 bit - Server 
* For this I followed these instructions https://iroboteducation.github.io/create3_docs/setup/pi4/
* I am using ROS2 Galactic
* I used echo "export RMW_IMPLEMENTATION=rmw_fastrtps_cpp" >> ~/.bashrc
* The dance demo worked in this case!

Hope this helps narrow the issue!

justinIRBT, 2023-03-07:
Which firmware is the robot running?  If you are trying Humble on Ubuntu you should also be running the latest humble firmware on the robot (H.1.0 as of the time of this post).  Until you see topics / services from your terminal, the dance script won’t work, so we should start by trying to get that working.  Can you confirm the robot firmware you are using in the robot webserver and while there double check the RMW the robot is configured to.  Can you post screenshots of the two webserver pages.

bluepra, 2023-03-07:
[image]
[image]

This is what I have. It looks like my firmware was for galactic. I will update to H.1.0 and report back! Thank you!

bluepra, 2023-03-07:
[image]
[image]
[image]

Ok so I updated the firmware to be H.1.0, and I also changed the RMW to be CycloneDDS.

This is what my .bashrc looks like on my Pi (Ubuntu 22.04):
```
source /opt/ros/humble/setup.bash
source ~/Robotics/create3_examples_ws/install/local_setup.sh
source /usr/share/colcon_argcomplete/hook/colcon-argcomplete.bash
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
```

Seems like my pi isnt able to communicate with the Roomba via my USB-C connection. any thoughts on this?

bluepra, 2023-03-07:
I disconnected ALL USB devices from my pi, except for the USB C connection to the Roomba and reran ``lsusb```
[image]

shamlian, 2023-03-07:
Two things: 
1) Did you configure your Raspberry Pi as an Ethernet gadget (editing /boot/config.txt and /boot/cmdline.txt), plugging the USB-C from the Create 3 Adapter Board into the USB-C on your Raspberry Pi? 
2) Is the switch on the adapter board in the "USB" position (blue LED on the adapter board is off)?

bluepra, 2023-03-07:
1. This is what I get when I do ifconfig: ```usb0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.186.3  netmask 255.255.255.0  broadcast 192.168.186.255
        inet6 fe80::b4db:88ff:fe44:52d  prefixlen 64  scopeid 0x20<link>
        ether b6:db:88:44:05:2d  txqueuelen 1000  (Ethernet)
        RX packets 37  bytes 1900 (1.9 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 185  bytes 27284 (27.2 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0```
2. The switch on the adapter board is switched to USB and the color of the light is yellow only.
3. When I do ```ping 192.168.186.2``` I get successfully communication. I believe 192.168.186.2 is the default IP of the roomba according to https://iroboteducation.github.io/create3_docs/setup/pi4/

bluepra, 2023-03-07:
I also don't see /boot/config.txt and /boot/cmdline.txt files on my pi in the /boot folder.

shamlian, 2023-03-07:
Given that you are able to ping the robot, my concern seems to be a red herring. I was responding to your `lsusb` call, but now I'm not sure whether or not `lsusb` can be used for upstream USB devices (in this case, the robot is upstream of the Pi).

shamlian, 2023-03-07:
What did you mean when you said, "Seems like my pi isnt able to communicate with the Roomba via my USB-C connection."? You say you can ping the robot. What happens if you try to access the webserver with something like `curl`?

bluepra, 2023-03-07:
It was just a guess that the pi couldn't communicate with the roomba. Does the ```ping 192.168.186.2``` working mean that the pi CAN communicate via the usb-c connection with the roomba?

```ros2 topic list``` also gets stuck with no output. 

Do you want me to launch the Roomba's web server and then try and connect via curl?

shamlian, 2023-03-07:
Please, don't guess, and please, give us full output of commands.
What does "gets stuck" mean? How long are you waiting? How do you stop the command?

bluepra, 2023-03-07:
It gets stuck for a long time (30 seconds plus) with no output. Control C doesn't work, and I had to use control Z to get access to the terminal again. Closing the terminal prompts me with an error saying that the process in the background would be killed too.

bluepra, 2023-03-07:
How do I access the webserver with curl?

shamlian, 2023-03-07:
Do you get a pile of html if you `curl 192.168.186.2/index.html` ?

carlsondc-ceva, 2023-03-07:
FWIW, I've had the ros2 topic/node utilities get unhappy when I'm monkeying around with interface settings. `ros2 node list --no-daemon` and `ros2 topic list --no-daemon` (or doing `ros2 daemon stop` before calling the utility) has occasionally seemed to unjam situations.

bluepra, 2023-03-07:
```curl 192.168.186.2/index.html``` gives some HTML code (I can post that if that's helpful).

shamlian, 2023-03-07:
No need; I think we can confirm that your network connection is working. I think the next thing to do is to take @carlsondc-ceva 's suggestion and bonk ROS 2 on your Pi in the head using `ros2 daemon stop` before trying `ros2 topic list` again.

shamlian, 2023-03-07:
Oh -- another thought -- if you're using Cyclone DDS, can we also see your XML configuration file?

bluepra, 2023-03-07:
[image]

I did ```ros2 daemon stop``` and then ```ros2 topic list``` and I get the same "stuck" behavior.

bluepra, 2023-03-07:
Where can I find the XML configuration file? I didn't create one myself at any point

shamlian, 2023-03-07:
If you didn't make one, it doesn't exist! :) That might be a problem source -- Cyclone in particular seems to need to be told which network interface to use. Stand by.

bluepra, 2023-03-07:
[image]

Just saw this, should I make a XML file?

shamlian, 2023-03-07:
Perhaps. Before you do that, without changing or unplugging anything, could you try, from the Pi, `curl -X POST http://192.168.186.2/api/reboot` to reboot the robot in place? Also, if you could post that logs.txt that I had you download before, that might be helpful.

bluepra, 2023-03-07:
[logs.txt](https://github.com/iRobotEducation/create3_examples/files/10913235/logs.txt)

I just did ```curl -X POST http://192.168.186.2/api/reboot``` and the robot did its restart chime

bluepra, 2023-03-07:
The ros2 node list command still doesn't work (same issue)

shamlian, 2023-03-07:
OK. Just to be 100% sure, after the robot has restarted, can you briefly try `ros2 topic list --no-daemon` to see if it hangs?

shamlian, 2023-03-07:
could you try _topics_ with also the --no-daemon flag?

bluepra, 2023-03-07:
> could you try _topics_ with also the --no-daemon flag?

What do you mean by this?

shamlian, 2023-03-07:
I meant the thing you did (you did it, but it didn't work. sad face). Might be time to try specifying the network interface using an XML file.

bluepra, 2023-03-07:
appreciate your help! Just let me know what I should put in the XML

shamlian, 2023-03-07:
try
```
<CycloneDDS>
   <Domain>
     <General>
        <NetworkInterfaceAddress>usb0</NetworkInterfaceAddress>
    </General>
   </Domain>
</CycloneDDS>
```
in a file with the name of your choosing, then set the environment variable 
`export CYCLONEDDS_URI=/path/to/the/xml/profile`
and then
`ros2 daemon stop`
and then
`ros2 topic list`

bluepra, 2023-03-07:
Created cyclone_usb.xml in the my ~ folder.
Added this to .bashrc ```export CYCLONEDDS_URI=/home/robotics_roomba/cyclone_usb.xml``` 

Stopped the ros2 daemon and then did ```ros2 topic list``` and i got this:
```1678217865.539523 [0]       ros2: config: //CycloneDDS/Domain/General: 'NetworkInterfaceAddress': deprecated element (/home/robotics_roomba/cyclone_usb.xml line 4)```

and nothing else...its stuck again, had to use Control-Z to get acess to terminal

bluepra, 2023-03-07:
Should I try switched to rmw_fastrtps_cpp

shamlian, 2023-03-07:
You're welcome to try -- make sure to switch both the robot and the Pi, and also restart the daemon after doing so.

shamlian, 2023-03-07:
I am concerned about the log, actually -- could you try reinstalling H.1.0?

bluepra, 2023-03-07:
So I swapped to fastrtps on the roomba
[image]

And on my pi's .bashrc I have ```export RMW_IMPLEMENTATION=rmw_fastrtps_cpp"```
I also commented the line that says ```export CYCLONEDDS_URI=/path/to/the/xml/profile```

I restarted my Pi, and killed the ROS2 daemon, and then did ``ros2 topic list``` and go this:

```
/parameter_events
/rosout
```

It didn't get stuck this time, but the roomba topics are missing unfortunately.

bluepra, 2023-03-07:
Should I go ahead and reinstall H.1.0 on the roomba? I am using this link https://iroboteducation.github.io/create3_docs/releases/h_1_0/ to download it btw.

shamlian, 2023-03-07:
What does `ros2 topic list --no-daemon --spin-time 10` produce? 
If this gets you nowhere, I would say it is worth trying to reflash the robot with H.1.0 again. There was an odd message in the log that makes me wonder if it fully updated.

bluepra, 2023-03-07:
With the cycloneDDS the ```ros2 topic list --no-daemon --spin-time 10``` would hang.

And with the fastrtps I just get:
```
/parameter_events
/rosout
```

I notice that https://iroboteducation.github.io/create3_docs/releases/h_1_0/ says to have 
[image]

Do I need to download/update my cycloneDDS on my pi?

bluepra, 2023-03-07:
In the meanwhile I will reinstall the H.1.0

shamlian, 2023-03-07:
I suspect you should not need to do anything to your ROS 2 install.

bluepra, 2023-03-07:
So I reinstalled the H.1.0 on the roomba, the while light was spinning, then a chime, then solid yellow for a few seconds, and then solid white.

Stilll got fastrtps on both the roomba and the pi

shamlian, 2023-03-07:
Trying to read between the lines here: you updated over the soft access point, and the robot took the update, but then because you haven't configured the robot's Wi-Fi, you can't access it from whatever computer you used to update it. Is the Pi still connected? Does `ping 192.168.186.2` get you packets back?

bluepra, 2023-03-07:
yes the ```ping 192.168.186.2``` still is able to transmit and receive packets.

Rebecca-iRobot, 2023-03-07:
Hi @bluepra! I am a learning designer at iRobot. I'd love to learn more about what you're specifically trying to do and how I can help :) Would you be up for scheduling a call to troubleshoot live? You can email us at education@irobot.com and I'll get in touch with you!

bluepra, 2023-03-07:
Yes I'll send that email over!

bluepra, 2023-03-09:
Can I please get a response to my email! I would like to get past this hurdle, so I can work on more interesting projects with the Roomba Create3.

Rebecca-iRobot, 2023-03-09:
Hi there! Thanks for your patience; we're a small team and have had a busy few days :)

bluepra, 2023-03-09:
No worries I understand, appreciate the help!

bluepra, 2023-03-17:
So I ended completely re-flashing Ubuntu 22.04 Desktop on my SD card, and re installing ROS2 Humble. I got the dance demo working and the Roomba topics to show up using FastRTPS. For whatever reasons CycloneDDS still doesn't work. 

One thing to note is that, occasionally the Roomba topics disappear. This is fixed by giving the Roomba a restart.

closed 2023-03-17 by bluepra