iRobotEducation/create3_examples #47: Cannot locate rosdep definition

raysu1027, 2023-04-14:
Hi 

I am step by step setup create3 but command **rosdep install --from-path src --ignore-src -yi** have some issues.

ros2@ros2-desktop:~/create3
/_examples_ws$ rosdep install --from-path src --ignore-src -yi
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
create3_teleop: Cannot locate rosdep definition for [teleop_twist_joy]
create3_lidar: Cannot locate rosdep definition for [slam_toolbox]

what can i do

alsora, 2023-04-14:
Hi,
what version of ROS 2 did you install and how did you do that?

raysu1027, 2023-04-14:
Hi alsora,
It's ROS 2 Galactic. I am follw this 
link. https://iroboteducation.github.io/create3_docs/setup/ubuntu2004/

[image]

alsora, 2023-04-14:
Those instructions are for a desktop (x86_64) build of ROS 2.
I think you are running on a Jetson Xavier, so those instruction are likely not correct.

raysu1027, 2023-04-14:
Hi alsora,
But create3 documentation can use on jenson.
like this https://iroboteducation.github.io/create3_docs/setup/jetson/
Am i missing something?

alsora, 2023-04-14:
Yes, but note that these instructions are different from those you linked before.
Did you follow the jetson specific instructions ?

And FYI, this problem is not rekated to Create 3 software, but rather "how to setup ROS 2 on NVidia jetson".

shamlian, 2023-04-14:
Also, to chime in here: Jetson support is via the community, and this is issue is not with Create 3 examples; I am moving this to a discussion, and we can continue it there with the community.