iRobotEducation/create3_examples #50: Lastest Lidar package broken

scottcandy34, 2024-01-26:


slowrunner, 2024-01-26:
@scottcandy34 what branch, in what way, on what fw, with what LIDAR, powered in what connection, with what SBC, with what OS?

Surely you can say more than "it's broken", no?

scottcandy34, 2024-01-26:
Sorry Reload I accidently created the issue before I could make a comment.

scottcandy34, 2024-01-26:
Ok I figured it out. Can you delete this line. 

https://github.com/iRobotEducation/create3_examples/blob/humble/create3_lidar_slam/launch/sensors_launch.py#L41 

The rplidar_composition does not like this and throws an error. Changing the name will not work either. It needs to be removed 

I can put in a pull request for this if you would like.

scottcandy34, 2024-01-26:
This is the output once that line gets removed.
```
gcc@ubuntu:~/ros2_ws$ ros2 launch create3_lidar sensors_launch.py
[INFO] [launch]: All log files can be found below /home/gcc/.ros/log/2024-01-26-14-45-06-242419-ubuntu-15054
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [static_transform_publisher-1]: process started with pid [15070]
[static_transform_publisher-1] [WARN] [1706298306.726169622] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-1] [INFO] [1706298306.767661462] [static_transform_publisher_2oAi9aUZ053DHeBh]: Spinning until stopped - publishing transform
[static_transform_publisher-1] translation: ('-0.012000', '0.000000', '0.144000')
[static_transform_publisher-1] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-1] from 'base_footprint' to 'laser_frame'
[INFO] [rplidar_composition-2]: process started with pid [15095]
[rplidar_composition-2] [INFO] [1706298308.776607173] [rplidar_node]: RPLidar running on ROS2 package rplidar_ros. RPLIDAR SDK Version:2.0.0
[rplidar_composition-2] [INFO] [1706298308.830661832] [rplidar_node]: RPLidar S/N: 8381EDF9C7E29BD1A7E39EF21352431B
[rplidar_composition-2] [INFO] [1706298308.830868867] [rplidar_node]: Firmware Ver: 1.29
[rplidar_composition-2] [INFO] [1706298308.830970236] [rplidar_node]: Hardware Rev: 7
[rplidar_composition-2] [INFO] [1706298308.882440884] [rplidar_node]: RPLidar health status : 0
[rplidar_composition-2] [INFO] [1706298308.882624345] [rplidar_node]: RPLidar health status : OK.
```

alain-lnpcrd, 2024-02-08:
Hi,
I've been tearing my hair out for over a month
1 lidar return on amazon.fr
1 more in preparation elsewhere
A great thank you

scottcandy34, 2024-02-08:
Should I just make a pull request to fix this or will someone else do that? 

It seams other people are having problems with this and are getting the wrong help since the errors are the same for the help that's offered.
https://github.com/iRobotEducation/create3_docs/discussions/513
https://github.com/iRobotEducation/create3_docs/discussions/520

It would be awful if more people run into this after pulling the latest package.

shamlian, 2024-02-08:
Thanks all, sorry for the delay. You've root-caused this correctly and I will fix it.

shamlian, 2024-02-08:
PR here: https://github.com/iRobotEducation/create3_examples/pull/51 . Once merged, this ticket will be closed.

scottcandy34, 2024-02-08:
Once again @shamlian your awesome thx.

shamlian, 2024-02-08:
That's very kind, thanks; you did all the work! 😅

closed 2024-02-08 by shamlian